oceanofthelost/InteractivePCB #4: iBOM not working when variant name include double quotes

dronecz, 2019-03-15:
Thanks for awesome script. 

I tried it and I did not see BOM, just header page. I debugged it with my friend and he found out, that my export contained double quotes in variant name of one library item. If he exported JSON, I could view it, If I did it no one was able to see BOM. I changed my variant name and both of us can see my exported BOM. 

I think the problem is in .js file, but I do not know JAVA so it is just wild guess.

oceanofthelost, 2019-03-17:
Could you provide an example output file that generates this. I think I have recreated the issue you mention, but I want to double check that what I get is the same as you.

dronecz, 2019-03-17:
Sure, here you have it. 

I think problem start at line 78 with "TACTILE_SWITCH_5X5_MICRO" 😉 

Thanks
[pcbdata_json.txt](https://github.com/oceanofthelost/InteractiveBOM/files/2975823/pcbdata_json.txt)

oceanofthelost, 2019-03-17:
Looking through this it seems that what is exported from Eagle side. It seems that the part TACTILE_SWITCH_5X5_MICRO in your schematic has a double quote in its VALUE field, and I believe in the one of the parts ATTRIBUTES as well. 

There also happens to be a second bug that this issue points out. My assumptions when porting iBOM to Eagle was that the attributes would all be unique, but for some reason you have two VALUE entries. Don't know how that happened in Eagle. Now this bug does not effect your application because the the value associated with VALUE, TACTILE_SWITCHE_5X5_MICRO is used for both entries. 

Attached you will find a minimum working example that almost matches what you get. 

[BUG0004.zip](https://github.com/oceanofthelost/InteractiveBOM/files/2975875/BUG0004.zip)

dronecz, 2019-03-18:
Yes, I already did that and I know, what I need to look for if I will have same problem in the future. I report this because someone else can have same issue and at least, will now how to make it work. 

Just to be clear, I changed just my ATTRIBUTES to make it work, so I´m not sure if double quotes i VALUE field is any problem.

oceanofthelost, 2019-03-19:
Fixed in commit 84720017a016ba750a36266c65750e66ed9e245b

closed 2019-03-19 by oceanofthelost