oceanofthelost/InteractivePCB #5: Feature request

dronecz, 2019-03-16:
I did play a bit with iBOM and for me, and hope for others, could be good to have this:

- **Quick search clear** : what about **X** at the end of the line which will clear value in search line? 
- **Color change for highlighted part**: if I check part as placed, would be possile to change color to greeen so I don´t have to worry about it anymore.
- **Hide placed parts**: if part is placed, hide the line with it so I will see just parts which have to be placed. I imagine this as **checkbox**, same as **Dark mode** in settings.  

I looked at JS file and CSS, but do not know, where to edit this. 

Thx 😉

oceanofthelost, 2019-03-17:
Ill look into this. JS and CSS are not my normal programming languages, But I do agree that these would be good additions.

oceanofthelost, 2019-03-21:
Color change of placed part currently implemented in 74213910db1b9b92ef256e4b44553e66b0b53fed, the current head of development commit.

oceanofthelost, 2019-03-23:
Added "Quick Search Clear" to filter bar. Added in commit e4a67119ba13dd7d75aafb8c59a01b3e02f4cdff.

oceanofthelost, 2019-03-24:
Added "Hide Placed Part" option to settings tab in commit e70603993a34e5a2570f58bcf404cf3e361e50a2

oceanofthelost, 2019-03-24:
### Feature Request Summary

1. Quick Search Clear: 74213910db1b9b92ef256e4b44553e66b0b53fed
2. Color Change For Highlighted Part: e4a67119ba13dd7d75aafb8c59a01b3e02f4cdff
3. Hide Placed Parts: e70603993a34e5a2570f58bcf404cf3e361e50a2

closed 2019-03-24 by oceanofthelost