oceanofthelost/InteractivePCB #11: Open file with JSON data

dronecz, 2019-04-02:
Could be possible to integrate file openning? 

I mostly export everything project related to project folder so I would love to open JSON data file from this folder. After that, I can export BOM jsut once for every project and donť have to export it everytime when I want work on other project. 

Thanks for reply

tablatronix, 2019-11-28:
Could maybe do it with filereader, but opening any local file is going to depend on browser and security settings, getting 1 solution to work across everything might not be possible.

https://www.html5rocks.com/en/tutorials/file/dndfiles/

oceanofthelost, 2019-12-07:
This will be incorporated into version 2 of this tool. Currently its a ways off since the entire front end is being redone in Python.

oceanofthelost, 2020-08-11:
I would like to add this to the HTML version of this program. Don't now what would be required to do that.

oceanofthelost, 2021-12-16:
Button added in 17b64da9b42fdcba9ad57fda4b8a8156b678d694. 

Redrawing canvas and tables currently not working.

oceanofthelost, 2022-01-11:
Added in 6b2cb964aed87f4c99cd43f4e6394d1b2ae0fb19. 

Support will be in V3 only.

closed 2022-01-11 by oceanofthelost