oceanofthelost/InteractivePCB #15: Missing silk names

tablatronix, 2019-11-28:
My pcb is missing the names labels, any idea why?

sample

```JSON
"bom": {
	"both": [
		[1,"","1X02_ROUND",["J3"],"VALUE","","B"],
		[1,"","1X06_1.27MM",["PROGRAM"],"NAME;VALUE","PROGRAM;","F"],
		[1,"","1X10_ROUND",["JP2"],"NAME;SPICEPREFIX;VALUE","JP2;X;","F"],
		[1,"","1X10_ROUND70",["JP1"],"NAME;VALUE","JP1;","F"],
		[1,"100nf","C0603",["C7"],"MF;MPN;NAME;OC_NEWARK;VALUE",";;C7;unknown;100nf","F"],
		[1,"10uf","C0805",["C1"],"MF;MPN;NAME;OC_NEWARK;VALUE",";;C1;unknown;10uf","F"],
		[1,"100nf","C0805",["C6"],"MF;MPN;NAME;OC_NEWARK;VALUE",";;C6;unknown;100nf","F"],
		[1,"100nf/1uf","C0805",["C8"],"MF;MPN;NAME;OC_NEWARK;VALUE",";;C8;unknown;100nf/1uf","F"],
		[1,"10uf","C0805K",["C2"],"MF;MPN;NAME;OC_NEWARK;VALUE",";;C2;unknown;10uf","F"],
		[1,"100uf","C3225",["C3"],"MF;MPN;NAME;OC_NEWARK;VALUE",";;C3;unknown;100uf","F"],
		[1,"100uf 12V","C3225",["C4"],"MF;MPN;NAME;OC_NEWARK;VALUE",";;C4;unknown;100uf 12V","B"],
```

oceanofthelost, 2019-11-28:
Th names and labels are not showing up on the PCB within iBOM?

tablatronix, 2019-11-29:
[image]
[image]
[image]

tablatronix, 2019-11-29:
Some silk shows, not all, also missing is all my tnames, reference names.

tablatronix, 2019-11-29:
I am using staging, and are capable of debugging and development, let me know where to look and I can help.

oceanofthelost, 2019-11-29:
Silkscreen must be on either the TPLACE or the BPLACE layers.

Now on the EagleCAD ULP side ([ibom.ulp](https://github.com/oceanofthelost/InteractiveBOM/blob/master/Plugins/Eagle/ibom.ulp)) there is a lot of room for expansion to extract the silkscreen data.

The function that extracts silkscreen data is GenerateSilkscreen on line 307. 

The problem that you are seeing I believe is all within this function.

tablatronix, 2019-11-29:
Was this by design? Why not NAMES? Silk is on place, and names..

oceanofthelost, 2019-11-29:
This is by design due to EagleCAD stating that silkscreen should be on the TPLACE and BPLACE layers, as per the layer reference article "[Every Layer in Autodesk EAGLE and What You Need to Know About Them](https://www.autodesk.com/products/eagle/blog/every-layer-explained-autodesk-eagle/)".

Looking at what issue you ran into I now see what the problem your having. I should add procesing of the following layers for the silkscreen

1. TPLACE/BPLACE
2. TVALUE/BVALUES
3. TNAMES/BNAMES

tablatronix, 2019-11-29:
Yeah that is wrong, Silkscreen in eagle is on 

T/BNAMES - part ^names
T/BVALUES - part ^values
T/BPLACE - part graphics, placeholders, special graphics etc.

oceanofthelost, 2019-11-29:
I would say its not wrong, more that clear definitions of how layers work within EagleCAD is not clearly defined.  The referenced post by Autodesk is from 2017-08-18, which was version 8.3 of the software. 

Following is a specification for the enhancement.  

1. Process EagleCAD PCB with ULP and extract part value entries. All entries will be entered as a segment type into the JSON file under silkscreen section. 
 2. Process EagleCAD PCB with ULP and extract part name entries. All entries will be entered as a segment type into the JSON file under silkscreen section. 
3. New type added to silkscreen option in JSON file to distinguish between name, value, and place, silkscreen. 
4. User settings added to iBOM to control display of silkscreen part names and silkscreen part values.

tablatronix, 2019-11-29:
I suppose, since it is entirely optional which layers you use in cam, that should be probably explained more in that, but the default CAM includes place and names, if your board is tiny, you would exclude names, of it was a giant kit board you might include values. 

So yeah its subjective. That is why I was thinking it might be best to have a custom layer for this and a ulp to combine them, or add options to the ulp.

tablatronix, 2020-08-05:
What ever happened with this? Is this still in development for python or whatever somewhere?

tablatronix, 2020-08-05:
the workaround I use for now is the copy-silk-screen ulp

SeanAlling, 2020-08-05:
I no longer use EagleCAD and have switched to only using KiCAD due to the direction EagleCAD is moving. Because of this I haven't had the motivation to put further development into the tool for EagleCAD.

oceanofthelost, 2020-08-05:
I no longer use EagleCAD and have switched to only using KiCAD due to the direction EagleCAD is moving. Because of this I haven't had the motivation to put further development into the tool for EagleCAD.

The Python work that I started is located in the repo under the branch "ConvertingToPython" branch.

oceanofthelost, 2020-08-10:
@tablatronix Would you be interested in being a collaborator of this project? This way you can make changes directly to the project.

tablatronix, 2020-08-10:
Sure like I need another oss project... rofl

For real though I would definitely contribute. I plan on fixing some stuff for my own use anyway

oceanofthelost, 2020-08-13:
I pushed a first draft  that address the text on the different layers not being displays. The changes can be found in the latest commit (5fc04df34f6b14d1e2c5fdb591de4f6b86599468).

Right now the ULP just dumps all visible text on the place, value, and name layers to the silkscreen layer in iBOM. There is no facility in the ULP or iBOM to select which of the three layers will be displayed. This is to be added.

oceanofthelost, 2020-10-15:
Started implementing the fix for this in ff5db2b7fc4bac889731755136519849acf44207. 

New file format will have specification for user defined layers. In iBOM layers will be able to be toggled by the user to show/hide layers.

oceanofthelost, 2020-12-20:
Implemented in commit 0721d90c97282d90037aa0c09dd6eec5c373ac1e. 

This commit is located on branch NewFileFormat.

closed 2020-12-20 by oceanofthelost