oceanofthelost/InteractivePCB #16: Merge/fork from openscopeproject

tablatronix, 2019-11-29:
How much of the GUI code did you have to modify? Can we continue to merge in features from there? Maybe turn GUI folder into submodule ?

It would be great to see if they can be maintained together.

oceanofthelost, 2019-11-29:
I will have to look at the current state of my fork and the original design by openscopeproject. 

I am planning to make my changes as a plugin to the original iBOM, but need to look into this some more.

tablatronix, 2019-11-29:
Looks good, I have peeked back and forth and see nothing major of concern, but ill keep an eye out for any major upgrades or bugfixes.

tolot27, 2020-01-08:
Can I support this development somehow? I've tried the branch ConvertingToPython but it did not work so far.
I don't know what to do with the `pcbdata.json` containing python code. Simply applying `generate_interactive_bom.py` to it does not work.

Maybe an extension like the [EasyEDA parser](https://github.com/openscopeproject/InteractiveHtmlBom/issues/95) is the way to go?

tablatronix, 2020-01-09:
ditto, also willing to beta test the python branch if needed.

PTS93, 2020-04-26:
Hows this looking? I think its hard to keep up with development as a fork, lots of stuff being added to the original project.

oceanofthelost, 2020-08-10:
There seems to be a lot of interest in this, I'm able to start supporting this again. 

My goal is to have this program implemented in Python using wxWidgets. I'll upload instruction on how to use the Python branch. Then others can at least test drive that one. 

In the mean time Ill see about integrating the changes in the  original Openscope version into this one.

tablatronix, 2020-08-11:
You are redoing the front end in python also? How will we be able to use that Across devices and over the network like html?

oceanofthelost, 2020-08-11:
A majority of the tooling I have developed for circuit automation is already in python. I was planning to  see if I could do this in python.

Ive opened issue #34 to investigate alternatives.  If I'm going to put additional work into this it makes more sense to leverage the work here instead of doing a complete rewrite.

PTS93, 2020-08-12:
If I understand the original setup correctly they added a sort of module support so you just have to write some part in python to parts the board files (see their EasyEDA support). This sounds more sensible than to make a completely new thing given how actively maintained the original project is and the amount of features it receives.

tablatronix, 2020-08-12:
Yeah I really like the lightweight portable html front end, I launch a python http server in a project folder and then use the front end on my ipad on my bench or my lab computer.

oceanofthelost, 2020-08-12:
I'm leaning more towards  sticking with HTML now since I have more experience in HTML/JS/CSS compared to earlier. 

@tablatronix you are right in that the HTML makes it a lot easier to use it on any device compared to other solutions.

tablatronix, 2020-08-13:
Just curious what you meant by “most of the tooling for circuit automation” means? Do you have plans for this beyond just a clone of ibom? Not sure I am clear on what you were referencing.

oceanofthelost, 2020-08-13:
I have developed multiple tools for internal (unreleased) development dealing with circuit automation and PCB analysis. The tools do things such as statistical analysis, trace impedance, trace resistance calculations, and more. 

What I am planning to do is to use iBOM as the Graphical front end and take the tools that i've developed, mainly as python programs, and get them to work with iBOM.

oceanofthelost, 2021-12-31:
Closing issue as with all development done to iPCB and where I want to take it, I don't think it will be able ti integrate back into openscopeproject. This project is essentially a hardfork and will not be backwards compatible.

closed 2021-12-31 by oceanofthelost