oceanofthelost/InteractivePCB #17: Feature request

SebDominguez, 2019-12-03:
Thanks you very much for this plugin.
This is the features that I wish this add-on could do

### Include non PCB parts
I'm looking for a BOM ULP that is able to generate a BOM including non PCB parts.
At first I was using BOM-EX but with this ULP I had to maintain a text database and I can't find it anything useful. The only great thing with BOM-EX is that on your part you can add an attribute `BOM` with the value `INCLUDED` or `EXCLUDED`.
This is useful as in many project I do have non PCB / Board parts that I want in my BOM (stands off, screws, power supply, fans, socketed ic, cable harness for instance).

Every BOM ULP for eagle just ignore parts that doesn't have footprint for the BOM and this just doesn't make any sens for me.
 
### Export to PDF / CSV
It would be so nice to be able to generate a printable friendly version of the ibom webview or to export the output to CSV so it can be imported into another piece if software

### Curved board outlines
I don't know if this is due to the last version of eagle that I'm using but all my curved shapes looks like the Tesla Cybertruck.

[image]
[image]

oceanofthelost, 2019-12-03:
Each feature that you are asking for are things that I would like to add to iBOM. Ill add them to feature request list. 

It looks like I might be able to start working on these in a couple of weeks.

oceanofthelost, 2019-12-07:
Started work on these items. All three feature requests will be integrated into version 2, but it is a about a moth out since the entire front end is being redone.

oceanofthelost, 2019-12-07:
I split this request into three feature requests in order to track each request individually.

closed 2019-12-07 by oceanofthelost