oceanofthelost/InteractivePCB #18: Add Documentation

oceanofthelost, 2019-12-03:
Currently there is no README or any documentation on how any of the program works. It will be useful for everyone if one was created.

tablatronix, 2020-08-08:
hmm how does the filter and attributes fields work? I always forget, is it delimited , wildcards? regex?

oceanofthelost, 2020-08-10:
Attribute properties are broken into two parts. Eagle exports the two ';' delimited lists. The first is attributes names, and the second is attributes values. 

these list are then read into pcb.js

https://github.com/oceanofthelost/InteractiveBOM/blob/e70603993a34e5a2570f58bcf404cf3e361e50a2/GUI/src/pcb.js#L58-L59

At this time split(';') converts the values from being ';' delimited  and generates an array of the values. 

It is assumed that ```attributeNames```  and ```attributeNames```  have the same number of items, if not then this would be an error. 

For example 
```'json
{"NAME;VALUE;","POWER;;"}
```

The first string represents attributeNames and the second string represents attributeNames. After calling split
```
attributeNames  = ["NAME", "VALUE"]
attributeValues = ["POWER", ""]
```

See Issue #27 , for more info on how filtering works.

tablatronix, 2021-12-29:
- [ ] Add a placeholder or an example for filtering syntax

oceanofthelost, 2021-12-29:
Information on removing BOM entry can be found in issue #69 as this fixed the feature not working.

PR #107 (2022-08-04): WIP: added README template 
Borrowed [template from Embedded Artistry](https://github.com/embeddedartistry/templates/blob/master/oss_docs/README.md) and started to edit

## What GitHub issue does this PR apply to?

Resolves #18 
Resolves #58 

## What changed and why?

Adding README with versioning and usage documentation. Starting from Embedded Artistry templates and editing as appropriate. 

## Checklist

   

- [ ] I claimed any associated issue(s) and they are not someone else's
- [ ] I have looked at documentation to ensure I made any revisions correctly
- [ ] I tested my changes locally  [... cut]
closed 2022-08-04 by oceanofthelost