oceanofthelost/InteractivePCB #25: Automatically hide parts

quotschmacher, 2020-06-23:
if i have the option checked to hide placed parts, all parts marked as placed are hidden. but if i check a new part, it does not hide. but that is what i would expect. instead i have to diasalbe the option to hide parts and enable it again.

(tested on chorme Version 83.0.4103.106 (Offizieller Build) (64-Bit))

tablatronix, 2020-08-08:
It looks like the bom is only updated on load or filter after changing checkboxes.

you add this to make it update on toggle..

```
function createCheckboxChangeHandler(checkbox, bomentry) {
    return function() 
    {
        if(bomentry.checkboxes.get(checkbox))
        {
            bomentry.checkboxes.set(checkbox,false);
            globalData.writeStorage("checkbox" + "_" + checkbox.toLowerCase() + "_" + bomentry.reference, "false");
        }
        else
        {
            bomentry.checkboxes.set(checkbox,true);
             globalData.writeStorage("checkbox" + "_" + checkbox.toLowerCase() + "_" + bomentry.reference, "true");
        }
       populateBomTable(); // <-- add
       
    }
}
```

oceanofthelost, 2020-08-10:
Lets add this change to the actual release and then close this out.

oceanofthelost, 2020-08-10:
I will have some time tonight, I can integrate this fix into the program.

oceanofthelost, 2020-08-11:
Resolved in commit ca1ceea97f144e6c5fd02855f9c2581705a65fa8. Will be released in next release V1.6.

closed 2020-08-11 by oceanofthelost