oceanofthelost/InteractivePCB #27: Add attr:value and ref:value pair filtering

tablatronix, 2020-08-08:
It looks like `Remove BOM Entries` only filter on entire attribute keys... `part.attributes.has(i)`
It would be nice to  add ref:value and attr:value pairs to this also.

```
reference:"JP1"
NAME:JP1
```

ref could probably easily be added by using `entryMatches(part)`, it seems to already do some lifting

For some reason I thought this could be done already , but I am not seeing anything in the code to do it...

oceanofthelost, 2020-08-10:
What you should look at is the following

https://github.com/oceanofthelost/InteractiveBOM/blob/e70603993a34e5a2570f58bcf404cf3e361e50a2/GUI/src/pcb.js#L80-L97

What this function does is take a bomtable and some filtering function. Apply the filtering function to every element, and then return the resulting bomtable. 

You can see an instance of this at
https://github.com/oceanofthelost/InteractiveBOM/blob/e70603993a34e5a2570f58bcf404cf3e361e50a2/GUI/src/ibom.js#L282-L295

Here the BOM table is being filtered to only show parts for the front or the back or the following which filters bases on user specified attributes

https://github.com/oceanofthelost/InteractiveBOM/blob/e70603993a34e5a2570f58bcf404cf3e361e50a2/GUI/src/ibom.js#L339-L351

To get the feature you want, a new filtering function needs to be defined. The filtered function should return true if the part should be removed from the viewable bom tables or false if it should remain. 

I think to get what you want, change the following line 

https://github.com/oceanofthelost/InteractiveBOM/blob/e70603993a34e5a2570f58bcf404cf3e361e50a2/GUI/src/ibom.js#L345

So that the passed filter function takes the reference, value, and attributes, or whatever else.

tablatronix, 2021-12-29:
I worked on this a bit and lost it, I think the kicad version has some advanced filtering like regex, it would be nice to match this, maybe I am mistaken. Going to revisit this.