oceanofthelost/InteractivePCB #28: Cannot click on some targets

tablatronix, 2020-08-08:
Can you give me some tips on where to debug this, Ill try to fix it.

I have several components that I cannot click on, but they are there and can be selected from the table.

oceanofthelost, 2020-08-10:
A little background, each part has a bounding box. When a part is selected the red (or green) box around the part is the parts bounding box. The bounding box is also used during  mouse clicked events to select a part. For exampled clicking on a part, and the click happened within a bounding box, then that part will be selected. 

The issue here I believe is due to the calculated bounding box that is done within Eagle. There are two locations that need to be looked at

1. EagleCAD ULP

https://github.com/oceanofthelost/InteractiveBOM/blob/eb841f75a04b6f48541f0d1d8f2c26d61b261147/Plugins/Eagle/ibom.ulp#L449-L451

Last I remember is that the bounding box calculation here is not the best and whats here was my best attempt to get this to work. This requires though that the Eagle part have a well defined bounding box which for this case is the package definition. Internally Eagle saves the bounding box for a parts as a UL_AREA data member. 

From the Eagle ULP manual

```
A UL_AREA is an abstract object which gives information about the area covered by an object. For a UL_DEVICE, UL_PACKAGE and UL_SYMBOL the area is defined as the surrounding rectangle of the object definition in the library, so even if e.g. a UL_PACKAGE is derived from a UL_ELEMENT, the package's area will not reflect the elements offset within the board. 
```

That last part I think is what causes the issues.

2. render.js
The following function renders the bounding box on the screen 
 
https://github.com/oceanofthelost/InteractiveBOM/blob/74213910db1b9b92ef256e4b44553e66b0b53fed/GUI/src/render.js#L211

https://github.com/oceanofthelost/InteractiveBOM/blob/74213910db1b9b92ef256e4b44553e66b0b53fed/GUI/src/render.js#L21-L226

3. render.js
https://github.com/oceanofthelost/InteractiveBOM/blob/74213910db1b9b92ef256e4b44553e66b0b53fed/GUI/src/render.js#L488-504

This function takes the location of a clicked mouse and if its within a part bounding box then selects that part. Now this could also be wrong. I don't remember how I cam up with these calculations. 

To solve this problem, I think that 1, and 3 needs to be addressed. The bounding box calculation within Eagle I think could be refined, as Ive seen some bounding boxes that it outputs for a part that do not reflect the actual area that the part package takes up. Then couple this with refining the calculations in 3 for selecting the clicked bounding box.

tablatronix, 2020-08-10:
Ok thanks, I was looking for that and a debug mode for making the bounding and hits visible

oceanofthelost, 2020-08-11:
The example pcbdata.json file for the Arduino mega shows this problem. Look at XIOH, and and XIOL.

oceanofthelost, 2020-08-11:
New option "Debug Mode" implemented in f8cc7c3d6221c3a089493164ac80a509ba41e67d. When set will display bounding box for all parts. 

Selected part will obey the placed/not placed color scheme.

oceanofthelost, 2020-12-31:
Release candidate V2.0-RC1 (b4efb650efd07c363dd9d428a95a31f8c678309e)  uses a new method for calculating the bounding box for a part.  Now the UL_AREA construct is used which gives the total bounding box for a part AND its labels. This can mean that the bounding box for the bounding box for the part is larger than the actual package size. 

Using UL_AREA means that the bounding box will at least never be smaller than the package area.

closed 2020-12-31 by oceanofthelost