oceanofthelost/InteractivePCB #34: investigate using Python or Electron for V2

oceanofthelost, 2020-08-11:
What method is there to not have to rewrite this whole application. What tools are aviviable that can leverage all of the work that been done but have a local app that can be used. 

The minimum requirements are 
1. User selectable pcbdata.json file. Should be able to open some kind of file explorer and navigate to the location of the file, 
1. Support exporting a pdf or BOM
1. Support exporting a pdf or svg of the rendered PCB

oceanofthelost, 2021-04-07:
In order to avoid rewriting the entire project, and to reuse software that has already been written, Python is no longer an option. 

Looking at how to integrate iPCB into Electron. This shall allow all developed code to be reused.

tablatronix, 2021-04-07:
Why not python ?

oceanofthelost, 2021-04-07:
There are a couple of reasons
1. Needing to rewrite most of the application using Python. 
2. Drawing performance
3. Needing to support multiple languages. 

I still am planning to make the app something native and testing with Python I was not able to get the same flexibility as what I currently have.

Python could be used to complement pieces of the system but the main rendering, and user interaction I shall keep with as HTLM/CSS/JS.