oceanofthelost/InteractivePCB #39: Reloading messes up placed data when selecting Combine values

Craftplorer, 2020-08-20:
When selecting multiple parts as placed and than reloading the page causes different parts to be checked as placed.

Steps to reproduce:
1: Select Combine values
2: Mark part/multiple parts as placed
3: Reload
Now different parts are marked as placed.

I suspect that the way or order Combine values and the "placed" data is loaded/executed causes this problem. 

Hope I was able to explain the problem clearly. Otherwise please contect me.

tablatronix, 2020-08-20:
Ill try to reproduce

Craftplorer, 2020-08-24:
I only testet the release 1.6 version. The master version already has fixed this bug. 
Seems related to https://github.com/oceanofthelost/InteractiveBOM/commit/17c216e209647d1cb242857e28514bf8e4335fd1

closed 2020-08-24 by Craftplorer

oceanofthelost, 2020-08-25:
This was addressed with Issue #8.

I'm currently working on the next release (V1.7). There a few more items that need to be tackled before its ready for release.