oceanofthelost/InteractivePCB #40: Fusion 360 support

Crazor, 2020-09-07:
Does this work with Fusion 360 Electronics Designs? It's essentially Eagle 9 repackaged into the Fusion GUI. I tried to run the ULP from a board file and got a pcbdata.json, but what do I do with that? I've used the KiCAD version before and expected an HTML file...

tablatronix, 2020-09-07:
The html loads the json file, you will need to put this in the folder with the html or, copy the html stuff to a new location with the json

Crazor, 2020-09-14:
Interesting, I'm pretty sure I tried exactly that. Of course, it is working now as you said ;)
Regarding Fusion 360: On first glance, it works just fine. However, the PCB outline I sketched in the 3D PCB and pushed back to the layout did not export correctly. Instead, I get a polygon with straight lines instead of arcs. I will investigate if this is an issue with arcs created in the 3D design and pushed back to the 2D or if this is an issue with old-style arcs as well.

oceanofthelost, 2020-10-15:
The current release version only supports lines, the development version at 03be01cc60d0d9656686928f9b8f506f931f17ba supports lines and arcs.

The issue in the current release is two fold, the developed EagleCAD ULP does not support exporting arcs and render.js does not support drawing arcs. Both issues are fixed in 03be01cc60d0d9656686928f9b8f506f931f17ba.

oceanofthelost, 2020-12-31:
Release candidate V2.0-RC1 (b4efb650efd07c363dd9d428a95a31f8c678309e) overhauled the types that can be rendered. This should allow rendering arcs, lines, and polygons. Unfortunately Autodesk has not released information on how Splines are implemented, so InteractivePCB is unable to render them for the time being.

oceanofthelost, 2021-12-29:
Need to address differences in ULP between Fusion360 Eagle and legacy EagleCAD. See Issue #47 for more information on depreciated ULP loop functions and data members.