oceanofthelost/InteractivePCB #46: 'Combine values' not combining all parts in V2.0-RC1

zchris9, 2021-01-07:
'Combine values' does not combine all parts with the same value if they are not in a single block of ordered reference designators. See figure:

[image]

oceanofthelost, 2021-12-16:
Issue stems from 

```js
GetBOMCombinedValues()
```

This function assumes the BOM table is sorted when combining values. When items are not sorted the above behavior is seen. 

Fix will be to update function to first sort parts by value and then combine.

oceanofthelost, 2022-01-11:
Resolved in V3. 

Sort is performed on part values and then displayed on BOM. Fix will not be back ported to V2.

closed 2022-01-11 by oceanofthelost