oceanofthelost/InteractivePCB #47: Fusion 360: polygons loop is deprecated

Crazor, 2021-01-12:
When I run the 2.0-RC1 ULP from within a Fusion 360 electronics design, I get the following error message:

`ibom.ulp(610): polygons loop is deprecated. Use polyPours polyShapes polyCutouts instead.`

Script execution is halted and the json file is incomplete.

oceanofthelost, 2021-01-12:
Thank you for pointing this out. 

One of the issues I am running into with Autodesk is that they haven't provided an updated ULP document. The latest document I  can get doesn't even mention polyShapes or polyCutouts. 

For now I wont be able fix this issue until Auotdesk releases an updated ULP document.

tablatronix, 2021-12-29:
These are located in fusion360 docs, at some point these will outdate eagle, as there has been no updates to eagle and probably will not be.

eg. 
https://help.autodesk.com/view/fusion360/ENU/?guid=ECD-ULP-POLYPOUR