oceanofthelost/InteractivePCB #49: IBOM with large PCB doesn´t work

Laminim, 2021-01-23:
There are some PCBs where the iBOM.html won´t show any details. I used the latest version and created a new pcbdata.json, but it just outputs an empty html. Please refer to the files attached. Using Eagle 9.6.2

[iBOM.zip](https://github.com/oceanofthelost/InteractivePCB/files/5860902/iBOM.zip)

oceanofthelost, 2021-01-23:
The issue seems to be part U10, TL072. This has the attribute DATA_SHEET which is assigned a value of 

```
"http://www.ti.com/lit/ds/symlink/tl084a.pdf"> Data sheet </a>
```

[iPCB.zip](https://github.com/oceanofthelost/InteractivePCB/files/5861145/iPCB.zip)

It looks like that iPCB does not currently work with parts that have attributes that are embedded links. I have attached a copy of a fixed pcbdata.json.

Laminim, 2021-01-23:
Thanks for the quick response and the corrected file. 

I successfully corrected this also for other projects. A simple workaround is to search for "http" or similar keywords and reduce the corresponding attributes line to something like this:

"attributes":{
				"name":"NAME",
				"value":"U10",
			},

tomasbubela, 2021-03-25:
I solved it by small modification of Eagle ULP file:
_original line:_
    `attributeValue += A.value + ";";`
_modificated_
    `attributeValue += removeDoubleQuote(A.value) + ";";`

oceanofthelost, 2021-12-16:
Resolved in 76b861e54dae7c6952e21cf02071433876ac67ca

closed 2021-12-16 by oceanofthelost