oceanofthelost/InteractivePCB #53: clicking line in bom persistently selects that part, not deselected when mouse moves off that line

smonk, 2021-11-17:
Hi,

I have been using this with eagle and it is very nice, so thanks.

One thing  I wondered is how hard would it be to make the bom list act like the pcb drawing wrt selecting parts? For example on the pcb view if i click on a part, it stays selected until i click another or move the mouse to the bom list. Would it be possible to add an option in settings so that the bom list also works with clicks? Right now it seems it just highlights whatever the mouse hovers over but it would be desirable to be able to click a bom line and have it highlight on the pcb, then stay highlighted until another line is clicked, not switch as soon as the mouse is not on that line.

oceanofthelost, 2021-11-17:
This behavior would actually not be to difficult. 

A new parameter will need to be added to the settings tab. 

The new parameter will then be used in https://github.com/oceanofthelost/InteractivePCB/blob/c73695dc5fc64fa6d9ee4099f35637c76eb23992/GUI/src/handlers_mouse.js#L67 to disable selection of a new parts. 

May need to adjust behavior of BOM entries as well. I would say fault behavior  (parameter off), the program behaves as it does now. Parameter on, clicking in BOM selects the BOM entry and highlights part on board, but clicking on PCB simply moves the PCB around.

smonk, 2021-11-17:
Ok, I may take a crack at it even, I know a small bit about js. 

thank you for your reply

oceanofthelost, 2021-11-17:
I would say use the parameter, _darkmode_ as an example and follow its implementation. 

Some javascript and HTML will need to be modified. The following are at least the minimum files to modify. 

1. ibom.html
2. htmlFunctions.js
3. ibom.js

The above would add the parameter to the settings drop down, create an event handler for enable disable, add the storage element so selection can be saved across page refresh. 

The last file to modify is 

1. handlers_mouse.js

Which would add the behavior you are looking for. 

Then use the include bat file to build the program to test and validate.

oceanofthelost, 2021-12-16:
I read what was being requested. I see now that what was requested is to click a BOM entry and have that  selection persist.

Commit e5d8dbd7d2b29d7f00d3b1693527374ce0474a63 changes the default behavior for the BOM table to now require clicking an entry to highlight the parts on the PCB. 

The commit is part of a new major update and has not been back ported, but to get the behavior in V2, a change in 

```js
function populateBomBody()
```
needs to be changed. 

**Before**
```js
 tr.onmousemove = handler;
 globalData.pushHighlightHandlers({
     id: tr.id,
     handler: handler,
     refs: references
 });
```

**After**
```js
 tr.onclick= handler;
 globalData.pushHighlightHandlers({
     id: tr.id,
     handler: handler,
     refs: references
 });
```

Next release will make onclick the default behavior.

closed 2021-12-16 by oceanofthelost

smonk, 2021-12-16:
hi,

thank you for the update, I will check it out.

We are using this utility a lot at work, it is much simpler for my tech that is not familiar with eagle to locate parts on the circuit boards, very well done.

oceanofthelost, 2021-12-16:
Will also need to incorporates another change as well. Commit d98f37c0c3d7355da5783c9edaac643a2b71d01c fixes an issue that is more predominate when using clicking.

This is also a single line change.

I will try and create a release with these changes.

oceanofthelost, 2021-12-16:
In Release V2.1 now available.