oceanofthelost/InteractivePCB #55: Add versioning and schema metadata

tablatronix, 2021-12-28:
I did not see any schema or versioning info in the html ui or the json, it would be useful I think to add such.

I tried opening some old json data and could not figure out how to read it.. etc

tablatronix, 2021-12-28:
For example , i just updated all my sources to 2.1 and none of my pcbdatas load. So apparently there is a constant breaking change to the schemas or formats.

tablatronix, 2021-12-28:
Ok maybe this is a bug, i reran the json same issue, ill make a new bug

oceanofthelost, 2021-12-28:
Protocol version is added to json from the ULP using the function

```c++
void GenerateMetadata( void )
```
Specifically, the lines of interest are 

```c++
PrintKeyValue_KeyString("protocol_version", "1");
PrintKeyValue_KeyString("ecad", "EagleCAD");
```

These two lines define what I believe is being requested.

A breaking change was introduced in V2.0 and V2.1. PCB's will need to be regenerated using ULP. Specifically the commit that introduced breaking changes was 4a48e9d1ae3ed032239c18288d277e768d805234 which was the last commit before release of V2.0.

Fix for version number is now made in b8d288220b6a0ba9b2d67d866915c13541eaed12 which will be used for V3.0 of program.

closed 2021-12-28 by oceanofthelost

tablatronix, 2021-12-28:
The html app should show program version somewhere or an about.