oceanofthelost/InteractivePCB #56: v2.1 not working ?

tablatronix, 2021-12-28:
```
render_canvas.js:35 Uncaught TypeError: Cannot read properties of undefined (reading 'bounding_box')
    at recalcLayerScale (render_canvas.js:35)
    at Object.ResizeCanvas (render_canvas.js:155)
    at Object.initRender (render.js:304)
    at window.onload (ibom.js:1269)
```

Blank
[image]

tablatronix, 2021-12-28:
```
Error: Parse error on line 1:
var pcbdata = {	"me
^
Expecting 'STRING', 'NUMBER', 'NULL', 'TRUE', 'FALSE', '{', '[', got 'undefined'
````

tablatronix, 2021-12-28:
The format is totally wrong..

part.attributes.name.split?

        "parts":[
            { "name":"part1",
              "attributes": [
                    {"name1":"value1"},
                    {"name2":"value2"},
                    {"name3":"value3"},
                ],

oceanofthelost, 2021-12-28:
Updated examples so that all now run once downloaded. Examples were not updated with latest release. 

JSON files generate prior to V2.1 and V2.0 will also need to be regenerated as file format used by program has changed.

tablatronix, 2021-12-28:
I regenerated using the 2.x ulp and it still doesn't work. Not sure whats going on.

tablatronix, 2021-12-28:
also there is 
https://github.com/oceanofthelost/InteractivePCB/tree/master/Examples/FileFormat

Which has not been updated

tablatronix, 2021-12-28:
my data doesn't look anything like the examples, you sure your ulp assets are updated here?

```
var pcbdata =  {
    "metadata": {
        "protocol_version": 2.1,
        "ecad": "Eagle",
        "company": "",
        "project_name": "esp_basic_esp32_20",
        "revision": "",
        "date": "2021-12-28 16:08:33",
        "number_parts": {
            "top": 18,
            "bottom": 29
        }
    },
    "board": {
        "bounding_box": {
            "x0": -18.76899,
            "y0": -11.032,
            "x1": 44.433,
            "y1": 52.5334
        },
        "traces": [
            {
                "name": "3V3",
                "segments": [
                    {
                        "type": "line",
                        "layer": 1,
                        "x0": 4.634,
                        "y0": 31.242,
                        "x1": 4.581,
                        "y1": 31.295,
                        "width": 0.4064
                    },
                    {
                        "type": "line",
                        "layer": 1,
                        "x0": 9.017,
                        "y0": 31.242,
                        "x1": 4.634,
                        "y1": 31.242,
                        "width": 0.4064
                    },
                    {
                        "type": "line",
                        "layer": 16,
                        "x0": 7.262,
                        "y0": 22.6338,
                        "x1": 7.8486,
                        "y1": 22.0472,
                        "width": 0.4064
                    },
                    {
                        "type": "line",
                        "layer": 1,
                        "x0": 7.112,
                        "y0": 11.2119,
                        "x1": 7.4629,
```

tablatronix, 2021-12-28:
Same for examples, ulp must be wrong

oceanofthelost, 2021-12-28:
The ULP being developed in master will not work with V2.X. Extensive changes have been made to simplify JSON data and rendering. 

The snippet above is telling me that you are using the ipcb.ulp that is in the master branch and not the ULP attached to a the latest release, [V2.2-RC2](https://github.com/oceanofthelost/InteractivePCB/releases/tag/V2.2-RC2). 

The source files for the GUI are located in iPCB.zip and the EagleCAD ULP to use is ipcb.ulp are both attached.

tablatronix, 2021-12-28:
So what branch is that?
Is master not cutting edge ?

I see development 2.2...

oceanofthelost, 2021-12-28:
Master branch is cutting edge, may not work as expected. Currently master is representing Release V3.0. 

Development_2 is the branch used for V2 and making any needed bug fixes. 

Release is where I tag release commits. 

For stable UI, one should use the Release tab on Github or the release branch.

tablatronix, 2021-12-28:
github is messing with me. I was using my own fork and its not synching branches for some reason!
Working now!

Sorry about that, I didn't realize there was a v3!

closed 2021-12-28 by tablatronix

oceanofthelost, 2021-12-28:
There may of been an issue with how I pushed all branches from my end. I think I have it resolved. Glad that it is resolved now though. 

Version 3 is a major overhaul of the GUI front end and backend. I will upload some documentation on this when I get the time.