oceanofthelost/InteractivePCB #60: undefined vars

tablatronix, 2021-12-28:
Where do all these variables come from ?
Global attributes?
Pulling in description?

[image]

oceanofthelost, 2021-12-28:
Part of the issue is EagleCAD ULP is not outputting company name or revision to JSON file.

In addition the displaying of metadata was also using incorrect variable names. 

Commit e1238714e871ad73c4e8b3e2a85166c5f8fb2c1c fixes at least the header portion of this issue.

oceanofthelost, 2021-12-28:
The other issue, Footprint showing up as undefined is due to EagleCAD ULP not outputting that data to JSON.

tablatronix, 2021-12-28:
Is that expected/known issue, or something I can look into ?

oceanofthelost, 2021-12-28:
Known issue. To fix I think the easiest way currently would be to modify the [ipcb.ulp](https://github.com/oceanofthelost/InteractivePCB/blob/e1238714e871ad73c4e8b3e2a85166c5f8fb2c1c/Plugins/Eagle/ipcb.ulp#L727) to something like the following 

```c++
E.attributes(A)
{
    attributeName += A.name + ";";
    attributeValue += A.value + ";";
}

attributeName  += "FOOTPRINT" + ";";    // Append footprint attribute
attributeValue += XX_FOOTPRINT_VALUE_XX + ";"; // Append actual footprint for part

```
Where ```XX_FOOTPRINT_VALUE_XX ``` is changed to grab the actual footprint value.

oceanofthelost, 2022-01-07:
Resolved in V3. 

If a value is not provided, then "" (empty string) will be displayed instead of undefined.

closed 2022-01-07 by oceanofthelost