oceanofthelost/InteractivePCB #61: Part bounding box contains value extents

tablatronix, 2021-12-28:
How can this be fixed, is this a limitation in the eagle ULP or eagle itself?

The parts highlights contain a bounding box coords defined by footprint and value, any way to fix this ?

[image]
[image]

oceanofthelost, 2021-12-28:
Related to issue #28. Validate UL_AREA is being used correctly in ULP.

oceanofthelost, 2021-12-29:
Code that calculates bounding box for part in ipcb.ulp

```c++
void PrintPackageBoundingBox(UL_ELEMENT E)
{
     PrintTabs();
    // Print the edge_box segment to the output file
    printf("\"bounding_box\":{");
    printf("\"x0\":%5.5f,", u2mm(E.package.area.x1));
    printf("\"y0\":%5.5f,", u2mm(E.package.area.y1));
    printf("\"x1\":%5.5f,", u2mm(E.package.area.x2));
    printf("\"y1\":%5.5f" , u2mm(E.package.area.y2));
    printf("},");
}
```

Will need to investigate why E.package.area is not giving correct bounding box.

tablatronix, 2021-12-29:
Thanks, it looks like  
"UL_AREA is an abstract object" so it contains unsmashed (undetached) texts and values

Seems that in order to work around this, would require a bit of looping and dimensional math to either loop texts() and undimension them from area, or loop other objects and vice versa.. seems like a PITA..

oceanofthelost, 2021-12-29:
From UL_Element, the center point for a part is given. 

Now the main challenge would be to iterate of UL_PACKAGE (sub-element of UL_ELEMENT) and find the actual width and length of the part. Then that data can be used along with the origin point to calculate the bounding box. 

Like you said it sounds like it will be a PITA. Its better than the original issue in #28 where the bounding box was smaller than the part.

tablatronix, 2021-12-29:
Yeah but it has to be calculated from UL_FOOTPRINT (legacy UL_PACKAGE) now..

<h2 id="loop-members" style="color: rgb(60, 60, 60); margin: 30px 0px 15px; font-size: 18px; font-weight: 500; font-family: ArtifaktElement, Arial, Helvetica, sans-serif; font-style: normal; font-variant-ligatures: normal; font-variant-caps: normal; letter-spacing: normal; orphans: 2; text-align: start; text-indent: 0px; text-transform: none; white-space: normal; widows: 2; word-spacing: 0px; -webkit-text-stroke-width: 0px; text-decoration-thickness: initial; text-decoration-style: initial; text-decoration-color: initial;">Loop members</h2><div class="ui-table-wrap" style="overflow: auto; max-width: 100%; color: rgb(60, 60, 60); font-family: ArtifaktElement, Arial, Helvetica, sans-serif; font-size: 14px; font-style: normal; font-variant-ligatures: normal; font-variant-caps: normal; font-weight: 500; letter-spacing: normal; orphans: 2; text-align: start; text-indent: 0px; text-transform: none; white-space: normal; widows: 2; word-spacing: 0px; -webkit-text-stroke-width: 0px; text-decoration-thickness: initial; text-decoration-style: initial; text-decoration-color: initial;">

  |  
-- | --
circles() | UL_CIRCLE
contacts() | UL_CONTACT
dimensions() | UL_DIMENSION
frames() | UL_FRAME
holes() | UL_HOLE
polycutouts() | UL_POLYCUTOUT
polyshapes() | UL_POLYSHAPE (see note)
rectangles() | UL_RECTANGLE
texts() | UL_TEXT (see note)
wires() | UL_WIRE (see note)

</div>

commit b52c9877 (2022-08-07): Created better method for calculating part bounding box.

Bounding boxes in EagleCAD can be defined with UL_Area, but this also includes
exploded text location for part with the bounding box of UL_AREA.

Now looking at the pins position and pin size, a bounding box is derived from
this information.

Closes #61
closed 2022-08-07 by oceanofthelost