oceanofthelost/InteractivePCB #67: Option to disable attr columns

tablatronix, 2021-12-29:
I think you can add attributes via the options to the table, but how can you remove stuff, say the footprint column now..
Maybe use the same method to allow removal special syntax or escape wrap ? (_remove)

oceanofthelost, 2021-12-29:
Commit 5ae172ed48bec65cc4d4b0daf2130b6fb1211b82 now allows user to specify if the footprint column should be shown. 

At a minimum, the only columns that cannot be removed are Value and Reference. Placed column is displayed by default but can be removed under the settings menu. 

@tablatronix, does 5ae172ed48bec65cc4d4b0daf2130b6fb1211b82 meet the intent of what you were looking for?

closed 2022-01-03 by oceanofthelost

oceanofthelost, 2022-01-03:
Added to release [V2.3.0](https://github.com/oceanofthelost/InteractivePCB/releases/tag/V2.3.0)