oceanofthelost/InteractivePCB #68: Filtering not working ? dev 2.x

tablatronix, 2021-12-29:
Nothing seems to show up when filtering, all rows disappear..

oceanofthelost, 2021-12-29:
This was due to a coding mistake in JS. Sometime before commit 5ae172ed48bec65cc4d4b0daf2130b6fb1211b82 the following was used in footprints

```js
 this.foorptint  = footprint;
```

but the above should of been 

```js
    this.footprint  = footprint;
```

The spelling mistake caused footprint to be undefined when called and since no code was written in the filter to deal wit undefined variables the filter function would hard crash. 

Commit 5ae172ed48bec65cc4d4b0daf2130b6fb1211b82 which addressed Issue #67 for removing footprint also fixed this issue.

closed 2022-01-03 by oceanofthelost

oceanofthelost, 2022-01-03:
Added to release [V2.3.0](https://github.com/oceanofthelost/InteractivePCB/releases/tag/V2.3.0)