oceanofthelost/InteractivePCB #78: Add fullscreen button

tablatronix, 2021-12-29:
Add a software fullscreen toggle, useful for mobile devices.

tablatronix, 2021-12-29:
I think this works but It has been a while , you might have to save bookmark to open fullscreen in mobile safari. Will test options

oceanofthelost, 2021-12-30:
I will have to research this. I am not familiar with how this can be done within HTML/JS

oceanofthelost, 2021-12-31:
[W3 Schools](https://www.w3schools.com/howto/howto_js_fullscreen.asp) has a tutorial that covers how to make something full screen using an onscreen button. 

My idea would be to add a button and a configuration parameter. Config parameter would indicate if program should start in full screen or not. Button would toggle between normal and full screen.

tablatronix, 2021-12-31:
Oh i have done this before on apps, I just need to look at it again and test it

oceanofthelost, 2021-12-31:
Button added to V3.X.X-Alpha (9f8974ca9627892c7d7b13297d2a9f4a3e445abc). Will backport button to V2. 

Logic is not included yet.

oceanofthelost, 2021-12-31:
Feature added in commits:

V2: 58147003ff3570bae0577ef1292cd2d83ee6be26
V3: 28e8ea1559646633e3489f8809853920ff60a10d

closed 2021-12-31 by oceanofthelost

oceanofthelost, 2022-01-03:
Added to release [V2.3.0](https://github.com/oceanofthelost/InteractivePCB/releases/tag/V2.3.0)