oceanofthelost/InteractivePCB #80: updated list high light behavior bug?

smonk, 2021-12-30:
Hello, regarding the mouse click on BOM to highlight a part. 

When you click on a part it highlights it and moving the mouse over other parts in the list does not change what part is highlighted (this is good). But there is an issue if you move the mouse to the board view, then back to the bom area (without clicking anything). When you return to the bom area the highlight around the part on the board disappears even though it is still highlighted in the bom list.

To recreate the bug:

Click  a line in the bom. See that the part is highlighted in the board view.

Move the mouse curser to the board view area. The part is still highlighted.

Move the mouse curser now back to the bom list. You should see the part is still highlighted in the bom, but the highlight from teh board view has disappeared.

I assume this is not intended so mention it here as a bug.

thanks

smonk, 2021-12-30:
i would also add that in a previous version clicking on the part in teh board view woudl highlight it in both board and bom. now it seems clicking on the part on the board view will highlight it in the bom but no highlight is drawn on teh board. this seems possibly related.

oceanofthelost, 2021-12-31:
Behavior shown was incorrect and is a bug. This bug was introduced when switching when switching from hover over BOM element to click selection element (Issue #53).

A mouse event was still enabled tat would force a redraw of the screen when mouse moved from the canvas back to the BOM. Both issues discussed were related.

Commit 30cee24f708bb7f8967c7249528ed3e62a4e8461 removes the mouse event which resolved the bug found.

closed 2021-12-31 by oceanofthelost

oceanofthelost, 2022-01-03:
Added to release [V2.3.0](https://github.com/oceanofthelost/InteractivePCB/releases/tag/V2.3.0)