oceanofthelost/InteractivePCB #98: Device names on tname and bname layer do not seem to export correctly anymore with ulp (also tplace/bplace)

smonk, 2022-01-17:
After trying out the newer release I seem to lose tnames and bnames associated with a device. Miscellaneous text on the board that has been put on tnames and bnames layer does export, However none of the names of any of the devices show up when those layers are turned on in the ipcb.html page. Previous versions did not seem to have this issue so i am just wondering if it is something to do with the way the new version of the ulp works.

It also seems tplace and bplace associated with the device also do not work properly anymore.

I have attached an image of the board in eagle showing that for certain at least "C4" is on bnames, and you can see none of the names of the parts show up in ipcb. You can also see in the ipcb screen grab some text "sftm ver 0.3 ..." is drawn, this text IS on bnames but it is not associated with a device, it was just generated using the text command in eagle. So it seems some of bnames layer worked, but not when its associated with a device.

[image]
[image]

smonk, 2022-01-17:
just to update, i looked more at the .ulp from an older version i had downloaded which worked and it does seem a lot of the function "PrintLayer" has been removed. In the newest version i don't think it works because when you use the texts() member of UL_BOARD object, it will not give you the text strings of the individual elements. This is silly but its how eagle ulp is organized. I simply copy and pasted the PrintLayer function from the older version into the new one and it seems to work fine now. I do not know the exact version of the older file but i have attached it here.
[ibom.txt](https://github.com/oceanofthelost/InteractivePCB/files/7884970/ibom.txt)

oceanofthelost, 2022-01-21:
Issue due to ULP not iterating over part loop variables for text and wires. 

Issue resolved in two commits:
1. tname/bname: 00c5b4fac128a5aa59d6d38517d01b09f09ea336
2. tplace/bplace: 7894bedcae76c6cd88b4a36f5627068828799a76

Released in [V2.3.1](https://github.com/oceanofthelost/InteractivePCB/releases/tag/V2.3.1)

closed 2022-01-21 by oceanofthelost