oceanofthelost/InteractivePCB #111: Allegro Support and Layers

ankithjv, 2023-06-12:
Does this integrate the inner layers of the pcb as well? As I know the InteractiveHTMLBom is for the front and back layers only. 
Also will there be integration for Allegro brd files as well?

oceanofthelost, 2023-06-12:
Currently evaluating if project can be used moving forward as EagleCAD Linux is no longer supported.

Do you have a reference for Allegro PCB file format?

Develop has support for all layers and all layers can be toggled. The “views” support top down and bottom up, so one can see a view norma or mirrored.

oceanofthelost, 2023-06-12:
> I can provide you with a sample allegro brd file. I have done the json conversion to run it through the interactivehtmlbom, however the script to convert the allegro pcb to json stripped it down to just front and back layers.

If you can provide a board file and whatever used to convert a board file to work with iPCB, I can look into adding support.

Take a look at some of the example boards, specifically Simple_BRD, as this is used to validate all edge cases with rendering.

ankithjv, 2023-06-13:
So I currently have an allegro .brd file and also a json however the json only has the front and back layers as that was the only open source code I could find to json convert.  I will send both. Is there a way for me to privately send you the files?

ankithjv, 2023-06-14:
I can now get a json file with all the inner files actually thanks @juulzA. Is there a python script I can run this through instead of as a plug in for eaglecad?

oceanofthelost, 2023-06-14:
you can upload an example PCB here or create a pull request with one.

Ideally the board would not be something proprietary. Even just having a PCB like an arduino board in allegro will help.

ankithjv, 2023-06-14:
I cannot upload brd or json files here or on a pull request